ttlappalainen/NMEA2000_esp32 #3: NodeMCU ESP-32S + MCP2551 doesn't seem to work

virtuvas, 2018-11-27:
Timo,

tried to use a NodeMCU ESP-32S I recently bought.
connected it to an MCP2551 exactly as I have it running on a teensy 3.5 and 3.6
compiled properly (had a bit of an issue uploading the code to the chip and had to keep the IOO button depressed for 5-6secs to connect)
However, although serial monitor shows various debug print lines (so it's working) the rest of the NMEA2K bus doesn't see it, and no sentences as broadcasted.

I've wired it as you recommend (and even tried adding the def lines in my code). Any ideas?

cheers

V.

markingle, 2018-11-30:
Hello guys!  I hate to highjack this thread but I’m not sure where/who to reach out to.  I am very familiar with the ESP32 and I’m interested in integrating it into the NMEA network on my boat. What are the hardware options to integrate the ESP32 into a NMEA network?  My email is markandcandyingle@gmail.com

ttlappalainen, 2018-11-30:
Use MCP2562. With other alternatives people have had problems.

In MCP2551 document it says Vdd 4.5-5.5 V. So you have to power it with 5 V. But then you will have problem with Rx, which will feed also 5 V, but ESP32 is 3.3 V tolerant. 

That is why I have used MCP2562 tranceiver in my devices. That can be powered with 5V and with Vio (pin 5) you control io level. Vio can be down to 1.8 V, so it works also with low voltage CPUs.

virtuvas, 2018-11-30:
success!

Timo, I tried also MCP2562 and used pin5 to 3.3V didn't work.

Eventually the problem was that it didn't like:

#define ESP32_CAN_TX_PIN GPIO_NUM_16
#define ESP32_CAN_RX_PIN GPIO_NUM_4

I noticed that on the docs I found online the NodeMCU-32S they mention:
TX0 ->GPIO1
RX0 ->GPIO3
and
TX2 ->GPIO17
RX2 ->GPIO16

so changed to

#define ESP32_CAN_TX_PIN GPIO_NUM_17
#define ESP32_CAN_RX_PIN GPIO_NUM_16

and it worked straight away.
BTW, it did so with both 2551 (no current on pin5) and 2562 with 3V3 on pin5 so everything is fine!

You do have me worried though with the 2551 which I'm using with all my teensies. could I face problems later on, or since it's working now fine for over a year, just leave it as is?

cheers

V.

ttlappalainen, 2018-11-30:
There are different ESP32 boards, which causes confusing with right pins.

I wouldn't use MCP2551 with 3.3V Teensyes 3.5 and 3.6.

markingle, 2018-11-30:
Thank you gentlemen!  I am going to be using the Sparkfun ESP32 Thing.  Its pretty rugged.  I tried several times to burn it up!  :)