ttlappalainen/NMEA2000_esp32 #4: Tx and RX for the MCP2562

mchampio, 2020-05-22:
I just want to ask a really stupid question.
The MCP2562 and MCP2515 combination according to their data sheets use SPI (MOSI, MISO,SCK CLK)
In your code and examples you talk about Tx and Rx. Is that because your code does not use a Bus Controller and instead talks directly to the MCP2562

ttlappalainen, 2020-05-22:
Can you point, which code or example you mean.

mchampio, 2020-05-22:
in NMEA2000_esp32.h you have 
gpio_num_t     TxPin;	
gpio_num_t     RxPin;

ttlappalainen, 2020-05-22:
ESP32 has internal CAN controller and its tx/rx can be set almost any pin. Those pins you connect directly to the rx/tx to MCP2562. Also DUE and Teensies has internal CAN, but with fixed pins.

If you use MCP2515, then you use SPI.

mchampio, 2020-05-22:
Thank you for the quick and comprehensive reply

closed 2020-05-22 by mchampio