ttlappalainen/NMEA2000_esp32 #7: Cant get it working with wemos D1 mini ESP32 (clone)

muebau, 2021-12-02:
Hello,
I tried everything to get my code working and found that there is no transmission on the CAN bus at all.

After a while of testing I tried the examples "Temperature Monitor" and "Battery Monitor" as described in the readme.
I used a logic analyzer (sigrok, saleae clone) to make sure there is some output on IO16, IO4, IO5, RX(UART) just to make sure. There is no output at all except on RX of course.

The "Temperature Monitor" was changed to include:
```
#include <NMEA2000_CAN.h>  // This will automatically choose right CAN library and create suitable NMEA2000 object
#include <NMEA2000_esp32.h> // https://github.com/ttlappalainen/NMEA2000_esp32
```
and I uncommented the serial output.

The only thing I found was some error messages ~10 sec after power up. I avoided resets as there was another issue.

```
PGN 130310 send failed
32727 : Pri:5 PGN:130310 Source:22 Dest:255 Len:8 Data:1,C1,70,FF,FF,FF,FF,FF
32734, Temperature send ready
PGN 130312 send failed
35224 : Pri:5 PGN:130312 Source:22 Dest:255 Len:8 Data:1,1,4,7D,73,FF,FF,FF
PGN 130311 send failed
35224 : Pri:5 PGN:130311 Source:22 Dest:255 Len:8 Data:1,C4,7D,73,FF,7F,FF,FF
PGN 130310 send failed
35232 : Pri:5 PGN:130310 Source:22 Dest:255 Len:8 Data:1,C1,70,FF,FF,FF,FF,FF
35239, Temperature send ready
PGN 130312 send failed
37725 : Pri:5 PGN:130312 Source:22 Dest:255 Len:8 Data:1,1,4,7D,73,FF,FF,FF
PGN 130311 send failed
37725 : Pri:5 PGN:130311 Source:22 Dest:255 Len:8 Data:1,C4,7D,73,FF,7F,FF,FF
PGN 130310 send failed
37733 : Pri:5 PGN:130310 Source:22 Dest:255 Len:8 Data:1,C1,70,FF,FF,FF,FF,FF
37740, Temperature send ready
PGN 130312 send failed
40226 : Pri:5 PGN:130312 Source:22 Dest:255 Len:8 Data:1,1,4,7D,73,FF,FF,FF
PGN 130311 send failed
40226 : Pri:5 PGN:130311 Source:22 Dest:255 Len:8 Data:1,C4,7D,73,FF,7F,FF,FF
PGN 130310 send failed
40234 : Pri:5 PGN:130310 Source:22 Dest:255 Len:8 Data:1,C1,70,FF,FF,FF,FF,FF
```

Do I something wrong?
Does the lib need something more than a pure Wemos D1 mini ESP32 to test for output?

ttlappalainen, 2021-12-03:
Add #include <NMEA2000_esp32.h> does nothing on your code. #include <NMEA2000_CAN.h> adds it automatically.

The most likely you have problem with NMEA2000 bus connection. Have you read the document https://github.com/ttlappalainen/NMEA2000/blob/master/Documents/Connecting_hardware_to_NMEA2000.pdf?

muebau, 2021-12-06:
Thank you a lot. The document guided a way to find a lot of little mistakes.
The fact that the "send failed" may appear after a longer delay showed that all nodes (not the one tested) had a problem.

In the end it was a missing GND at the USB2CAN combined with a broken solder joint on one of the modules.

So in fact it was a problem on the bus only!

closed 2021-12-06 by muebau