ttlappalainen/NMEA2000_esp32 #8: Unable to send/receive after starting alone at a BUS

wellenvogel, 2022-03-17:
In my project I use this lib and the NMEA2000 lib for a converter NMEA0183/NMEA2000 on a nodemcu32 or AtomCAN or other ESP32 based boards.
Whenever I start up with some other devices on the bus being already active everything runs fine.
But when starting as the only bus device (and connecting other devices later on) the library will not send or receive any PGNS. Looking at the bus the device is still continously retransmitting.
I tried a couple of changes as proposed e.g. at https://www.esp32.com/viewtopic.php?t=5010 - but still not very successful. At least when powering off the other device and later on powering on again the same issue occured.
So finally I decided to go for a re-initialization of the CAN driver whenever the send queue remains full for 2 seconds (or nothing is send/received within 60s).
Not really a great solution - but seems to solve the problem.
Do you have any suggestions/ideas how to handle this (maybe more inline with the CAN specs)?
My patched Version is at https://github.com/wellenvogel/esp32-nmea2000/tree/master/lib/nmea2000esp32.

ttlappalainen, 2022-03-19:
I did the test, but could not repeat your problem. I started ESP32 alone on the bus and let it run about 3 min. My device status shows then NMEA2000 error. Then I connected other devices and ESP32 started to send data to the bus normally.

I checked also that my Github has same version I have.

wellenvogel, 2022-03-19:
Ok, thanks for testing again.
We have this issues in 2 set ups.
Both consisting of only 2 esp32 on the bus. Normally the second one being switched on works normal whereas the first one runs into this problem.
Maybe we have to cross check if the error does not occur if we use a different device as the second one coming up.