ttlappalainen/NMEA2000_esp32 #14: Sketches not compiling

Baumistlustig, 2023-05-20:
It does not matter what I do with the library, it always throws these errors. This one was with the ActisenseListener example. 

```
/home/johannes/.arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/xtensa-esp32s3-elf-g++ -DHAVE_CONFIG_H "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DUNITY_INCLUDE_CONFIG_H -DWITH_POSIX -D_GNU_SOURCE "-DIDF_VER=\"v4.4.4\"" -DESP_PLATFORM -D_POSIX_READER_WRITER_LOCKS -I/home/johannes/.arduino15/packages/esp32/hardware/esp32/2.0.9/tools/sdk/esp32s3/include/newlib/platform_include -I/home/johannes/.arduino15/packages/esp32/hardware/esp32/2.0.9/tools/sdk/esp32s3/include/freertos/include -I/home/johannes/.arduino15/packages/esp32/hardware/esp32/2.0.9/tools/sdk/esp32s3/include/freertos/include/esp_additions/freertos -I/home/johannes/.arduino15/packages/esp32/hardware/esp32/2.0.9/tools/sdk/esp32s3/include/freertos/port/xtensa/include -I/home/johannes/.arduino15/packages/esp32/hardware/esp32/2.0.9/tools/sdk/esp32s3/include/freertos/include/esp_additions -I/home/johannes/.arduino15/packages/esp32/hardware/esp32/2.0.9/tools/sdk/esp32s3/include/esp_hw_support/include -I/home/johannes/.arduino15/packages/esp32/hardware/esp32/2.0.9/tools/sdk/esp32s3/include/esp_hw_support/include/soc -I/home/johannes/.arduino15/packages/esp32/hardware/esp32/2.0.9/tools/sdk/esp32s3/include/esp_hw_support/include/soc/esp32s3 -I/home/johannes/.arduino15/packages/esp32/hardware/esp32/2.0.9/tools/sdk/esp32s3/include/esp_hw_support/port/esp32s3 -I/home/johannes/.arduino15/packages/esp32/hardware/esp32/2.0.9/tools/sdk/esp32s3/include/esp_hw_support/port/esp32s3/private_include -I/home/johannes/.arduino15/packages/esp32/hardware/esp32/2.0.9/tools/sdk/esp32s3/include/heap/include -I/home/johannes/.arduino15/packages/esp32/hardware/esp32/2.0.9/tools/sdk/esp32s3/include/log/include -I/home/johannes/.arduino15/packages/esp32/hardware/esp32/2.0.9/tools/sdk/esp32s3/include/lwip/include/apps -I/home/johannes/.arduino15/packages/esp32/hardware/esp32/2.0.9/tools/sdk/esp32s3/include/lwip/include/apps/sntp -I/home/johannes/.arduino15/packages/esp32/hardware/esp32/2.0.9/tools/sdk/esp32s3/include/lwip/lwip/src/include -I/home/johannes/.arduino15/packages/esp32/hardware/esp32/2.0.9/tools/sdk/esp32s3/include/lwip/port/esp32/include -I/home/johannes/.arduino15/packages/esp32/hardware/esp32/2.0.9/tools/sdk/esp32s3/include/lwip/port/esp32/include/arch -I/home/johannes/.arduino15/packages/esp32/hardware/esp32/2.0.9/tools/sdk/esp32s3/include/soc/include -I/home/johannes/.arduino15/packages/esp32/hardware/esp32/2.0.9/tools/sdk/esp32s3/include/soc/esp32s3 -I/home/johannes/.arduino15/packages/esp32/hardware/esp32/2.0.9/tools/sdk/esp32s3/include/soc/esp32s3/include -I/home/johannes/.arduino15/packages/esp32/hardware/esp32/2.0.9/tools/sdk/esp32s3/include/hal/esp32s3/include -I/home/johannes/.arduino15/packages/esp32/hardware/esp32/2.0.9/tools/sdk/esp32s3/include/hal/include -I/home/johannes/.arduino15/packages/esp32/hardware/esp32/2.0.9/tools/sdk/esp32s3/include/hal/platform_port/include -I/home/johannes/.arduino15/packages/esp32/hardware/esp32/2.0.9/tools/sdk/esp32s3/include/esp_rom/include -I/home/johannes/.arduino15/packages/esp32/hardware/esp32/2.0.9/tools/sdk/esp32s3/include/esp_rom/include/esp32s3 -I/home/johannes/.arduino15/packages/esp32/hardware/esp32/2.0.9/tools/sdk/esp32s3/include/esp_rom/esp32s3 -I/home/johannes/.arduino15/packages/esp32/hardware/esp32/2.0.9/tools/sdk/esp32s3/include/esp_common/include -I/home/johannes/.arduino15/packages/esp32/hardware/esp32/2.0.9/tools/sdk/esp32s3/include/esp_system/include -I/home/johannes/.arduino15/packages/esp32/hardware/esp32/2.0.9/tools/sdk/esp32s3/include/esp_system/port/soc -I/home/johannes/.arduino15/packages/esp32/hardware/esp32/2.0.9/tools/sdk/esp32s3/include/esp_system/port/public_compat -I/home/johannes/.arduino15/packages/esp32/hardware/esp32/2.0.9/tools/sdk/esp32s3/include/xtensa/include -I/home/johannes/.arduino15/packages/esp32/hardware/esp32/2.0.9/tools/sdk/esp32s3/include/xtensa/esp32s3/include -I/home/johannes/.arduino15/packages/esp32/hardware/esp32/2.0.9/tools/sdk/esp32s3/include/driver/include -I/home/johannes/.arduino15/packages/esp32/hardware/esp32/2.0.9/tools/sdk/esp32s3/include/driver/esp32s3/include -I/home/johannes/.arduino15/packages/esp32/hardware/esp32/2.0.9/tools/sdk/esp32s3/include/esp_pm/include -I/home/johannes/.arduino15/packages/esp32/hardware/esp32/2.0.9/tools/sdk/esp32s3/include/esp_ringbuf/include -I/home/johannes/.arduino15/packages/esp32/hardware/esp32/2.0.9/tools/sdk/esp32s3/include/efuse/include -I/home/johannes/.arduino15/packages/esp32/hardware/esp32/2.0.9/tools/sdk/esp32s3/include/efuse/esp32s3/include -I/home/johannes/.arduino15/packages/esp32/hardware/esp32/2.0.9/tools/sdk/esp32s3/include/vfs/include -I/home/johannes/.arduino15/packages/esp32/hardware/esp32/2.0.9/tools/sdk/esp32s3/include/esp_wifi/include -I/home/johannes/.arduino15/packages/esp32/hardware/esp32/2.0.9/tools/sdk/esp32s3/include/esp_event/include -I/home/johannes/.arduino15/packages/esp32/hardware/esp32/2.0.9/tools/sdk/esp32s3/include/esp_netif/include -I/home/johannes/.arduino15/packages/esp32/hardware/esp32/2.0.9/tools/sdk/esp32s3/include/esp_eth/include -I/home/johannes/.arduino15/packages/esp32/hardware/esp32/2.0.9/tools/sdk/esp32s3/include/tcpip_adapter/include -I/home/johannes/.arduino15/packages/esp32/hardware/esp32/2.0.9/tools/sdk/esp32s3/include/esp_phy/include -I/home/johannes/.arduino15/packages/esp32/hardware/esp32/2.0.9/tools/sdk/esp32s3/include/esp_phy/esp32s3/include -I/home/johannes/.arduino15/packages/esp32/hardware/esp32/2.0.9/tools/sdk/esp32s3/include/esp_ipc/include -I/home/johannes/.arduino15/packages/esp32/hardware/esp32/2.0.9/tools/sdk/esp32s3/include/app_trace/include -I/home/johannes/.arduino15/packages/esp32/hardware/esp32/2.0.9/tools/sdk/esp32s3/include/esp_timer/include -I/home/johannes/.arduino15/packages/esp32/hardware/esp32/2.0.9/tools/sdk/esp32s3/include/mbedtls/port/include -I/home/johannes/.arduino15/packages/esp32/hardware/esp32/2.0.9/tools/sdk/esp32s3/include/mbedtls/mbedtls/include -I/home/johannes/.arduino15/packages/esp32/hardware/esp32/2.0.9/tools/sdk/esp32s3/include/mbedtls/esp_crt_bundle/include -I/home/johannes/.arduino15/packages/esp32/hardware/esp32/2.0.9/tools/sdk/esp32s3/include/app_update/include -I/home/johannes/.arduino15/packages/esp32/hardware/esp32/2.0.9/tools/sdk/esp32s3/include/spi_flash/include -I/home/johannes/.arduino15/packages/esp32/hardware/esp32/2.0.9/tools/sdk/esp32s3/include/bootloader_support/include -I/home/johannes/.arduino15/packages/esp32/hardware/esp32/2.0.9/tools/sdk/esp32s3/include/nvs_flash/include -I/home/johannes/.arduino15/packages/esp32/hardware/esp32/2.0.9/tools/sdk/esp32s3/include/pthread/include -I/home/johannes/.arduino15/packages/esp32/hardware/esp32/2.0.9/tools/sdk/esp32s3/include/esp_gdbstub/include -I/home/johannes/.arduino15/packages/esp32/hardware/esp32/2.0.9/tools/sdk/esp32s3/include/esp_gdbstub/xtensa -I/home/johannes/.arduino15/packages/esp32/hardware/esp32/2.0.9/tools/sdk/esp32s3/include/esp_gdbstub/esp32s3 -I/home/johannes/.arduino15/packages/esp32/hardware/esp32/2.0.9/tools/sdk/esp32s3/include/espcoredump/include -I/home/johannes/.arduino15/packages/esp32/hardware/esp32/2.0.9/tools/sdk/esp32s3/include/espcoredump/include/port/xtensa -I/home/johannes/.arduino15/packages/esp32/hardware/esp32/2.0.9/tools/sdk/esp32s3/include/wpa_supplicant/include -I/home/johannes/.arduino15/packages/esp32/hardware/esp32/2.0.9/tools/sdk/esp32s3/include/wpa_supplicant/port/include -I/home/johannes/.arduino15/packages/esp32/hardware/esp32/2.0.9/tools/sdk/esp32s3/include/wpa_supplicant/esp_supplicant/include -I/home/johannes/.arduino15/packages/esp32/hardware/esp32/2.0.9/tools/sdk/esp32s3/include/ieee802154/include -I/home/johannes/.arduino15/packages/esp32/hardware/esp32/2.0.9/tools/sdk/esp32s3/include/console -I/home/johannes/.arduino15/packages/esp32/hardware/esp32/2.0.9/tools/sdk/esp32s3/include/asio/asio/asio/include -I/home/johannes/.arduino15/packages/esp32/hardware/esp32/2.0.9/tools/sdk/esp32s3/include/asio/port/include -I/home/johannes/.arduino15/packages/esp32/hardware/esp32/2.0.9/tools/sdk/esp32s3/include/bt/common/osi/include -I/home/johannes/.arduino15/packages/esp32/hardware/esp32/2.0.9/tools/sdk/esp32s3/include/bt/include/esp32s3/include -I/home/johannes/.arduino15/packages/esp32/hardware/esp32/2.0.9/tools/sdk/esp32s3/include/bt/common/api/include/api -I/home/johannes/.arduino15/packages/e [... cut ...]
                           ^~~~~~~~~~
/home/johannes/Arduino/libraries/NMEA2000_esp32/NMEA2000_esp32.cpp:200:27: note: suggested alternative: 'TWAI_TX_IDX'
     gpio_matrix_out(TxPin,CAN_TX_IDX,0,0);
                           ^~~~~~~~~~
                           TWAI_TX_IDX

Using library NMEA2000 at version 4.18.7 in folder: /home/johannes/Arduino/libraries/NMEA2000 
Using library NMEA2000_esp32 at version 1.0.3 in folder: /home/johannes/Arduino/libraries/NMEA2000_esp32 
exit status 1

Compilation error: exit status 1```

ttlappalainen, 2023-05-21:
Seems that there is no those required definitions for ESP32-S2 and ESP32-S3 boards, so it will not compile with those boards for now. Compiling for ESP32 works.

GWS65, 2023-05-21:
Hello all,

I see that you are using the newest ESP32 core v2.0.9 ... Please try the
same with an install that has v1 core of esp32. I have seen multiple errors
with core v2.0.x that went away using the old core version I think you can
uninstall it in the board manager. (in Arduino IDE).

Hope this solves your issue for now
GWS

ttlappalainen, 2023-05-21:
Does old core compile properly for S2 and S3?  v2.0.x compiles OK, if you use board with ESP32 chip.

On the other hand e.g., ESPAsyncWebserver is horrible slow with newest core at least in AP mode.

jiauka, 2023-06-12:
Try this library
https://github.com/jiauka/NMEA2000_esp32xx
It uses the espressif original  HAL twai calls. It works on ESP32, ESP32C3 and has been successfully tested on ESP32S2 by @cheli0. It should work on any ESP32 chip/module with CAN/TWAI interface

mljuba1, 2024-01-27:
Hi all. 
@jiauka 
Please can you or any body help me on explanation how to integrate https://github.com/jiauka/NMEA2000_esp32xx lib into ActisenseListenerSender example. I would like to use it with ESP32-S2-Saola-1 dev module.

Thank you in advance....

Code:
// Demo: NMEA2000 library. Bus listener and sender. 
//   Sends all bus data to serial in Actisense format.
//   Send all data received from serial in Actisense format to the N2kBus.
//   Use this e.g. with NMEA Simulator (see. http://www.kave.fi/Apps/index.html) to send simulated data to the bus.
//   I have plans to add receiving functionality and data forwarding to NMEA Simulator. Meanwhile you can define
//   other stream to different port so that you can send data with NMEA Simulator and listen it on other port with 
//   Actisense NMEA Reader.

//#define N2k_CAN_INT_PIN 21
#include <Arduino.h>
#include <N2kMsg.h>
#include <NMEA2000.h>
#include <NMEA2000_esp32xx.h>
#include <NMEA2000_CAN.h>
#include <ActisenseReader.h>

Error:
In file included from C:\Users\Milan\Desktop\ActisenseListenerSender_esp32s2\ActisenseListenerSender_esp32s2.ino:14:
c:\Users\Milan\Documents\Arduino\libraries\NMEA2000-master\src/NMEA2000_CAN.h:278:27: error: expected type-specifier before 'tNMEA2000_esp32'
 tNMEA2000 &NMEA2000=*(new tNMEA2000_esp32());
                           ^~~~~~~~~~~~~~~
c:\Users\Milan\Documents\Arduino\libraries\NMEA2000-master\src/NMEA2000_CAN.h:278:26: error: expected ')' before 'tNMEA2000_esp32'
 tNMEA2000 &NMEA2000=*(new tNMEA2000_esp32());
                      ~   ^~~~~~~~~~~~~~~~
                          )

Using library NMEA2000 at version 4.20.2 in folder: C:\Users\Milan\Documents\Arduino\libraries\NMEA2000-master 
Using library NMEA2000_esp32xx at version 0.0.1 in folder: C:\Users\Milan\Documents\Arduino\libraries\NMEA2000_esp32xx-main 
Using library NMEA2000_esp32 at version 1.0.3 in folder: C:\Users\Milan\Documents\Arduino\libraries\NMEA2000_esp32-master 
exit status 1

Compilation error: exit status 1

ttlappalainen, 2024-01-28:
I expect you have not read the document - who needs them. On https://ttlappalainen.github.io/NMEA2000/pg_lib_ref.html chapter "Inherited classes according to your Board" you would have found examples for forced "driver" usage for several boards. Extending examples to new "driver" you ca do:
```
#include <Arduino.h>
#include <N2kMsg.h>
#include <NMEA2000.h>
#define ESP32_CAN_TX_PIN 16
#define ESP32_CAN_RX_PIN 4
#include <NMEA2000_esp32xx.h>
#include <ActisenseReader.h>

tNMEA2000_esp32xx NMEA2000();
```

Constructor tNMEA2000_esp32xx has defines above as default values for tx/rx pin parameters. So as alternative you can also use

```
#include <Arduino.h>
#include <N2kMsg.h>
#include <NMEA2000.h>
#include <NMEA2000_esp32xx.h>
#include <ActisenseReader.h>

tNMEA2000_esp32xx NMEA2000(16,4);
```

mljuba1, 2024-01-28:
Hi Timo. Thank you so much for help and pointing me to great doc that you have created. Thank you.... 
I did how its described 

_//   other stream to different port so that you can send data with NMEA Simulator and listen it on other port with 
//   Actisense NMEA Reader.

//#define N2k_CAN_INT_PIN 21
#include <Arduino.h>
#include <N2kMsg.h>
#include <NMEA2000.h>
//#include <NMEA2000_CAN.h>
#include <NMEA2000_esp32xx.h>
#include <ActisenseReader.h>

tNMEA2000_esp32xx NMEA2000();
tActisenseReader ActisenseReader;

// Define READ_STREAM to port, where you write data from PC e.g. with NMEA Simulator.
#define READ_STREAM Serial       
// Define ForwardStream to port, what you listen on PC side. On Arduino Due you can use e.g. SerialUSB
#define FORWARD_STREAM Serial_ 

and now this is error msg in console:

C:\Users\Milan\Desktop\test2\test2.ino: In function 'void setup()':
test2:30:12: error: request for member 'SetN2kCANSendFrameBufSize' in 'NMEA2000', which is of non-class type 'tNMEA2000_esp32xx()'
   NMEA2000.SetN2kCANSendFrameBufSize(150);
            ^~~~~~~~~~~~~~~~~~~~~~~~~
test2:31:12: error: request for member 'SetN2kCANReceiveFrameBufSize' in 'NMEA2000', which is of non-class type 'tNMEA2000_esp32xx()'
   NMEA2000.SetN2kCANReceiveFrameBufSize(150);
            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
test2:35:12: error: request for member 'SetForwardStream' in 'NMEA2000', which is of non-class type 'tNMEA2000_esp32xx()'
   NMEA2000.SetForwardStream(ForwardStream);
            ^~~~~~~~~~~~~~~~
test2:36:12: error: request for member 'SetMode' in 'NMEA2000', which is of non-class type 'tNMEA2000_esp32xx()'
   NMEA2000.SetMode(tNMEA2000::N2km_ListenAndSend);
            ^~~~~~~
test2:38:43: error: request for member 'SetForwardOwnMessages' in 'NMEA2000', which is of non-class type 'tNMEA2000_esp32xx()'
   if (ReadStream==ForwardStream) NMEA2000.SetForwardOwnMessages(false); // If streams are same, do not echo own messages.
                                           ^~~~~~~~~~~~~~~~~~~~~
test2:40:12: error: request for member 'Open' in 'NMEA2000', which is of non-class type 'tNMEA2000_esp32xx()'
   NMEA2000.Open();
            ^~~~
C:\Users\Milan\Desktop\test2\test2.ino: In function 'void HandleStreamN2kMsg(const tN2kMsg&)':
test2:51:12: error: request for member 'SendMsg' in 'NMEA2000', which is of non-class type 'tNMEA2000_esp32xx()'
   NMEA2000.SendMsg(N2kMsg,-1);
            ^~~~~~~
C:\Users\Milan\Desktop\test2\test2.ino: In function 'void loop()':
test2:55:12: error: request for member 'ParseMessages' in 'NMEA2000', which is of non-class type 'tNMEA2000_esp32xx()'
   NMEA2000.ParseMessages();
            ^~~~~~~~~~~~~
exit status 1
request for member 'SetN2kCANSendFrameBufSize' in 'NMEA2000', which is of non-class type 'tNMEA2000_esp32xx()'

Can you maybe help on this one or this?

Best regards...

riotenn, 2024-01-28:
I had your same problem.  I am using the Arduino Nano ESP32.  This is the S3 version of the ESP32.
Follow the instructions for modifying NMEA2000_CAN.h (in the src tree of NMEA2000) found at this link:
https://github.com/ttlappalainen/NMEA2000/issues/348
If using the Arduino IDE, make sure you add only NMEA2000_esp32xx as the support library.  Do not add NMEA2000_esp32  *thus your imported libraries should be NMEA2000 and NMEA2000_esp32xx).
It is not necessary to modify any other code except perhaps the pins you are using for TX and RX.  The modification to CAN.h shown in the link takes care of including the xx library and instantiating the correct NMEA2000 object.  
One of the default pins is not available on the Nano ESP32-S3 so it is necessary to change this.
I used Timo's Temperature Monitor as my core development "sketch."   I just uncommented his pin defines at the top and defined the pins I was using.  
Note that I was never able to get the Actisense Reader to work.  I spent perhaps 1/2 a day reading all the documentation and poring over the code but no luck.  I gave up on that since I do not really need it for my application.  
I have been successful reading and writing to a N2K network I have set up in the lab (for example, reading GNSS data on the ESP and transmitting dynamic engine parameters to a Garmin MFD). 

Thanks, Timo, for this huge code base!!

ttlappalainen, 2024-01-28:
Sorry my mistake. If you do not define any parameters and just use default you have leave () away.
```
#include <Arduino.h>
#include <N2kMsg.h>
#include <NMEA2000.h>
#define ESP32_CAN_TX_PIN GPIO_NUM_16
#define ESP32_CAN_RX_PIN GPIO_NUM_4
#include <NMEA2000_esp32xx.h>
#include <ActisenseReader.h>

tNMEA2000_esp32xx NMEA2000;
```

Also pins must be define with names as above. If you give pins for constructor you use:

```
#include <Arduino.h>
#include <N2kMsg.h>
#include <NMEA2000.h>
#include <NMEA2000_esp32xx.h>
#include <ActisenseReader.h>

tNMEA2000_esp32xx NMEA2000(GPIO_NUM_16,GPIO_NUM_4);
```

In Temperature monitor if you define on setup

```
Serial.begin(115200);
NMEA2000.SetForwardStream(&Serial);
NMEA2000.EnableForward(true);
```
Messages will be forwarded to USB port with 115200 baud and you can read them with e.g., Actisense Reader.

mljuba1, 2024-01-28:
**@riotenn** 
** Thank you for replay. Please see below.** 

** @ttlappalainen**  
** Successfully compiled!! Thank you for help.** 

** Info for others with same issue:** 
** Board - ESP32-S2-Saola-1** 
** IDE     - Arduino 2.2.1** 
** peek inside of Actisense listener sender sketch** 

//   Actisense NMEA Reader.

#include <Arduino.h>
#include <N2kMsg.h>
#include <NMEA2000.h>
#define ESP32_CAN_TX_PIN GPIO_NUM_16
#define ESP32_CAN_RX_PIN GPIO_NUM_4
#include <NMEA2000_esp32xx.h>
#include <ActisenseReader.h>

tNMEA2000_esp32xx NMEA2000;
tActisenseReader ActisenseReader;

// Define READ_STREAM to port, where you write data from PC e.g. with NMEA Simulator.

** Thank you all guys!!** 
** One other question, can someone point me to example of receiving PGN 127502 - Switch Bank Control msg **and controlling GPIO? :-)**

ttlappalainen, 2024-01-29:
Again read documents. There is Set and parse funtions for 127502 on N2kMessages.h. Note that this works in a bit different way. Device, which will set switch bank switch sends 127502 one with only new status for defined switch. Other switches has to be set NA. 

127502 is also deprecated. One should use command group function 126208 for commanding switch states.

mljuba1, 2024-02-03:
Thank you for reply Timo. You been helping me a lot. I am not skilled programmer but enthusiast.
Actually I don't know how to use 126208.

 I am trying to build 16ch relay sw board for my boat that I can switch on off or toggle by timer from my Garmin. I have success setting switches to on-off with this example https://github.com/ando274/N2KRelays ,  but only for 8ch. As soon as I change to 16ch by fiddling with bankStatus = bankStatus | 0xffffffffff000000ULL; // set unused bits to "unavailable" thing start to get weird and I can not help to myself to correct them. On Garmin I get 16 switches displayed though and when I press either of first 8 (SW1 to SW8) that one goes on but also sw from 9 to 16 goes on as well and vice-versa. Can someone please help regarding this. Thank you so much.

ttlappalainen, 2024-02-04:
What do you try to do with bankStatus = bankStatus | 0xffffffffff000000ULL; ? If you have not set bankStatus before, your result is undefined. If you have set, then you force 16 topmost switches NA and 12 lowest unchanged.

I prefer to use functions N2kResetBinaryStatus, N2kSetStatusBinaryOnStatus and N2kGetStatusOnBinaryStatus for status bits handling. Note that you should command change only for status you want to change, not for all. E.g., if you want to turn on switch 12, you should do:
```
tN2kBinaryStatus BankStatus;
N2kResetBinaryStatus(BankStatus);
N2kSetStatusBinaryOnStatus (BankStatus,N2kOnOff_On,12);
SetN2kSwitchState(N2kMsg,0,BankStatus);  // Expected you have written SetN2kSwitchState
```
Then you should listen 127501 to read current status.

You can read how to use Command Group Function from document https://web.archive.org/web/20170609033039/http://www.nmea.org/Assets/20140109%20nmea-2000-corrigendum-tc201401031%20pgn%20126208.pdf
That is mentions on documentation first page on chapter References.