ttlappalainen/NMEA2000_esp32 #17: esp32c3

jiauka, 2023-06-07:
Hi all!
The not so new  esp32c3 does no work with this library.

I have successfully implemented a new library that uses the native "twai" library from espresiff, so far it works with esp32c3 -at least receiving-. And I guess it should work with any "esp32" device (S2, etc..)

I need to do further test in order to triple check that it works find on my boat.

The question

Is there any interest in such library?

jeremy-brody, 2023-06-07:
Yes please! I work with the esp32-s3.

If you post a branch, I'd be happy to test and verify on that platform.

ttlappalainen, 2023-06-08:
If it compiles on PlatformIO Espressif 3.5, please send me zip with email. I'll run tests, gives first impression does it work or not.

I can't go to newer than Espressif 3.5 since after that ESP32 is slow like worm in AP mode.

jiauka, 2023-06-08:
Hi I do use esp-idf V5, I have never worked on platfomio :(. Anyway I send you a ZIP, it is a WIP and TX has NOT been tested

jiauka, 2023-06-08:
Hi, Here it is the esp32c3 library.

Tested ONLY on esp idf 5 and RX only. I need to go to the boat to  grab
the  MFD and test it.

ALso I need to test it under Arduino IDE, and I'm not a fan of it :)

ttlappalainen, 2023-06-08:
Why do you use tasks for transmit and receive? As far as I understand twai documentation, you can define rx and tx queue sizes for twai_general_config_t. Then you simply call if ( twai_receive(&rx_msg, 0)==ESP_OK ) ...; for receive and twai_transmit(&tx_msg, 0); for sending. With tasks you waste extra 10k of memory.

jiauka, 2023-06-08:
Well, as I told it is WIP, and I saw on original NMEA2000_ESP that is uses interrupts and a kind of weird queu handling, I decided to implement a couple of task (like in espresif example) for RX, TX.

Probably it is much better not to use them. I will work on it this weekend.

BTW, I have commit a new version, there was a lock at TX task :(
It works at TX, at least at my desk with a temperatur monitor TX and a text Actisense RX

ttlappalainen, 2023-06-08:
Original NMEA2000_ESP uses queues (not weird), since it read/writes directly CAN controller. Twai driver does that internally.

jiauka, 2023-06-08:
Ok, I will rewrite it. It was a kind of quick and dirty test in order to make it work :). Once I test it with a true B&G Network, I will remove those queues and tasks

jiauka, 2023-06-09:
Hi again:

Lib working at least with ESP32 and ESP32c3. It uses the twai espressif HAL, so it should work on every esp32xx device with CAN/TWAI hardware.
Tested with a B&G network that has some ESP32  and STM32F103 NMEA2000 devices too, and a VOLVO PENTA engine gateway, almost useless in my sail boat :(.

https://github.com/jiauka/NMEA2000_esp32xx

An example using esp-idf is located at

https://github.com/jiauka/NMEA2000-windesp32xx-idf

ttlappalainen, 2023-06-10:
Ome hints:
- Remove tasks and related definitions. I do not see any use for them.
- Define static const char *const TAG = "TWAI_C3"; It locates your string to flash instead of ram.
- Use ESP_LOGD for info. Not ESP_LOGE.
- You could also remove speed variable and remove switch. NMEA2000 speed is fixed to 250k
- also #if 0 if unnecessary. Just use hand configuration and comment it as "force speed 250k"
- Also define
```
#if CORE_DEBUG_LEVEL >= ESP_LOG_DEBUG
static const char *const TAG = "TWAI_C3";
#endif
```
to avoid TAG totally

jiauka, 2023-06-10:
Thanks for the tips!

Done and tested on both esp32 and esp32c3 :) check git

jiauka, 2023-06-10:
git pushed right now

cheli0, 2023-06-10:
Hi, I made an NMEA2000 data viewer with an ESP32-2432S028 and arduino IDE. It works perfect!. Now I'n trying to port the code on a ESP32-4827S043 (bigger screen but ESP32S3.... I´m added and included the NMEA2000_esp32xx library to the arduino IDE, and select ESP32S3 developer module on IDE but don´t compiles... Should I do more changes to the code? Can you help me? compiling on a ESP32C2/C3 doesn´t work either....
Thanks in advance!!

jiauka, 2023-06-10:
It works fine with Arduino IDE and esp32c3..
I have done a quick and dirty hack at NMEA2000_CAN.h
`
....
#elif USE_N2K_CAN == USE_N2K_ESP32_CAN
#if CONFIG_IDF_TARGET_ESP32C3
#include <NMEA2000_esp32xx.h>       // https://github.com/jiauka/NMEA2000_esp32xx
tNMEA2000 &NMEA2000=*(new tNMEA2000_esp32xx());
#else
#include <NMEA2000_esp32.h>       // https://github.com/ttlappalainen/NMEA2000_esp32
tNMEA2000 &NMEA2000=*(new tNMEA2000_esp32());
#endif
#else  // Use USE_N2K_MCP_CAN
...`

For S3 should be similar. I only have esp32 and esp32c3, sorry

cheli0, 2023-06-10:
Compiles like a charm!! Thank you so much!
Just had to add an "*" in the sentences
#include <NMEA2000_esp32xx.h> // https://github.com/jiauka/NMEA2000_esp32xx
tNMEA2000 &NMEA2000=*****(new tNMEA2000_esp32xx());
#else
#include <NMEA2000_esp32.h> // https://github.com/ttlappalainen/NMEA2000_esp32
tNMEA2000 &NMEA2000=*****(new tNMEA2000_esp32());
Thanks again. Next issues will go to https://github.com/jiauka/NMEA2000_esp32xx

cheli0, 2023-06-10:
Compiles like a charm!! Thank you so much!
Just had to add an "*" in the sentences after the "="
#include <NMEA2000_esp32xx.h> // https://github.com/jiauka/NMEA2000_esp32xx
tNMEA2000 &NMEA2000=*(new tNMEA2000_esp32xx());
#else
#include <NMEA2000_esp32.h> // https://github.com/ttlappalainen/NMEA2000_esp32
tNMEA2000 &NMEA2000=*(new tNMEA2000_esp32());
Thanks again. Next issues will go to https://github.com/jiauka/NMEA2000_esp32xx

cheli0, 2023-06-10:
Compiles like a charm!! Thank you so much!
Just had to add an * in the sentences after the "="
#include <NMEA2000_esp32xx.h> // https://github.com/jiauka/NMEA2000_esp32xx
tNMEA2000 &NMEA2000=*(new tNMEA2000_esp32xx());
#else
#include <NMEA2000_esp32.h> // https://github.com/ttlappalainen/NMEA2000_esp32
tNMEA2000 &NMEA2000=*(new tNMEA2000_esp32());
Thanks again. Next issues will go to https://github.com/jiauka/NMEA2000_esp32xx

closed 2023-06-10 by jiauka