ttlappalainen/NMEA2000_esp32 #20: error when running DataDisplay2-esp32

MeY0u, 2024-04-05:
hello,
i am testing the libraries with my environment, i installed 2 libraries: NMEA2000_esp32 and NMEA2000.
after towards i ran code verify with simple gpio adjusments and i receive the following  error: 
```
c:\Users\WoWpr\OneDrive\Documents\Arduino\libraries\NMEA2000_esp32\NMEA2000_esp32.cpp:123:23: error: 'PERIPH_CAN_MODULE' was not declared in this scope
   periph_module_reset(PERIPH_CAN_MODULE);
```

i didnt find any mention of "PERIPH_CAN_MODULE" in the libraries, could you pleases navigate me towards its source?

ttlappalainen, 2024-04-05:
NMEA2000_esp32 driver does not wotk with -S models. You can try https://github.com/jiauka/NMEA2000_esp32xx/tree/main, but I have not yet fully tested it.

MeY0u, 2024-04-06:
Thank you for your reply, i will look into it

MeY0u, 2024-04-06:
> NMEA2000_esp32 driver does not wotk with -S models. You can try https://github.com/jiauka/NMEA2000_esp32xx/tree/main, but I have not yet fully tested it.

okay, so i have tested the your suggestion it seems to work. using the link and Display2 example code.
i read successfully compass_directio, SOG, depth but the Temperature shows 16504 which is weird and cant be.
also i read allot of "In Main Handler: <some number>" .
what does it mean?
also some examples doesnt work , only this seems to decode the incoming data.

ttlappalainen, 2024-04-08:
Display2 is just one example for showing how you could read data. It is not meant for anything finally usable. Other examples are showing different use of library and they does not decode data at all.

Temperature printing 16504 is caused by invalid handling of extented temperature range "not available" value. It should show "not available". I'll fix that some day.

Just read the code - "In Main Handler: " is printed on the HandleNMEA2000Msg, when it gets any message. So it prints that first for every message and then if message is known by sample, it prints also parsed data. Bus contains a lot of messages this example does not handle.