ttlappalainen/NMEA2000_esp32 #25: ESP32 new version S2 S3 C3

forty76, 2024-05-21:
Hello, thank you for your fantastic libraries

Unfortunately Arduino detects several errors when using the new ESP32 (S2,S3,C3). I'm currently trying
     sergei/NMEA2000_esp32_twai (https://github.com/sergei/NMEA2000_esp32_twai)

You could implement the right libraries for the various devices in the NMEA2000_CAN.h file.

#elif USE_N2K_CAN == USE_N2K_ESP32_CAN
   #if CONFIG_IDF_TARGET_ESP32S2
     #include <NMEA2000_esp32_twai.h>
     tNMEA2000 &NMEA2000=*(new NMEA2000_esp32_twai());
   #elif CONFIG_IDF_TARGET_ESP32S3
     #include <NMEA2000_esp32_twai.h>
     tNMEA2000 &NMEA2000=*(new NMEA2000_esp32_twai());
   #elif CONFIG_IDF_TARGET_ESP32
     #include <NMEA2000_esp32.h>
     tNMEA2000 &NMEA2000=*(new tNMEA2000_esp32());
   #else
     #error "Unrecognized ESP."
   #endif

ttlappalainen, 2024-05-22:
I won't, since it does not use properly my library.

unxs0, 2024-06-12:
> I won't, since it does not use properly my library.

Can we crowd fund the work needed to get esp32-s3 support?

ttlappalainen, 2024-06-17:
Do you need it personally or commercially? Why S3?

unxs0, 2024-06-18:
Hi Timo!

The few manufacturers of low cost boat usable AMOLED displays only use this MCU for products and/or on their development kit boards.

They are DWO and their client LilyGo.

https://www.lilygo.cc/products/t4-s3 (https://www.lilygo.cc/products/t4-s3)

https://www.dwo.net.cn/pr.jsp?_pp=0_316_1_-1 (https://www.dwo.net.cn/pr.jsp?_pp=0_316_1_-1)

Plus the only good quality low cost easy to use graphics library LVGL adds extra problems, interfering in some way with TWAI on the Expressif S3. This is possibly due to some error in the Espxx lib that at least compiles and runs (fails after some time when LVGL controls SPI) with your excellent NMEA2000 library. 

It is for an open source boating mini multifunction display project. And of course it should be completely libre software part of your esp32 extensión.

Cheers,
Gary

ibravo, 2024-06-19:
+1 for adding support to the Lilygo T Display S3 board. These are great boards that have an attached screen, thus able to display the data on the same board, making it great for a small footprint device.
I'm also using it for personal use, to attach to the NMEA2k network and display the results in the little screen.

skarlsson, 2024-10-23:
I hacked together something that you can steal or take some inspiration from. https://github.com/skarlsson/NMEA2000_twai. I used it successfully (last week) for 3 devices esp32, esp32s3 and espc3  (olimex card and xiao) using esp idf 5.x

ieb, 2024-11-13:
@ttlappalainen does https://github.com/skarlsson/NMEA2000_twai conform to how you would expect support for ESP32 C3,S3 with the twai library support to look ?  

It seems more direct and simpler than over variants that have been shared in PRs  and issues here, but looks quite different from the original ESP32 version as it uses the twai driver methods. (ie -ISR replaced by twai queues, + a task)

The only problem I have found so far is https://github.com/skarlsson/NMEA2000_twai/blob/main/NMEA2000_esp32.cpp#L94C3-L99C23 fails to compile and had to be replaced by 

                      twai_message_t message = {
                          .flags = 0x01,
                          .identifier = id,
                          .data_length_code = static_cast<uint8_t>(len > 8 ? 8 : len),
                          .data = {0}
                      };
 
 probably due to a missing compiler flag in the defaults. Cant see why, the code matches the header file.

 error was
 
                    .pio/libdeps/esp32-c3/NMEA2000_twai/NMEA2000_esp32.cpp:126:5: error: designator order for field 'twai_message_t::<unnamed union>::<unnamed struct>::rtr' does not match declaration order in 'twai_message_t'

@skarlsson  I will do a PR with the change and details on my setup.

I am targeting esp32-C3 from waveshare and ebay for the firmware for https://github.com/ieb/NMEA2000_Booklet. Why ESP32-C3 ? Because the pcb looks more elegant than the version using a ESP32 Wroom, and I needed to integrate a JDB BMS. Might regret that and revert.

ieb, 2024-11-13:
standard build command line is 

          riscv32-esp-elf-g++ -o .pio/build/esp32-c3/lib93f/NMEA2000_twai/NMEA2000_esp32.cpp.o \
              -c -march=rv32imc -std=gnu++11 -fexceptions -fno-rtti -Os -ffunction-sections \
              -fdata-sections -Wno-error=unused-function -Wno-error=unused-variable \
              -Wno-error=deprecated-declarations -Wno-unused-parameter -Wno-sign-compare \
              -ggdb -Wno-error=format= -nostartfiles -Wno-format -freorder-blocks \
              -Wwrite-strings -fstack-protector -fstrict-volatile-bitfields -Wno-error=unused-but-set-variable \
              -fno-jump-tables -fno-tree-switch-conversion -MMD -DPLATFORMIO=60113 \
              -DARDUINO_ESP32C3_DEV -DASYNCWEBSERVER_REGEX -DHAVE_CONFIG_H \
              -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DUNITY_INCLUDE_CONFIG_H \
              -DWITH_POSIX -D_GNU_SOURCE -DIDF_VER=\"v4.4.7-dirty\" -DESP_PLATFORM \
              -D_POSIX_READER_WRITER_LOCKS -DARDUINO_ARCH_ESP32 -DESP32 -DF_CPU=160000000L \
              -DARDUINO=10812 -DARDUINO_VARIANT=\"esp32c3\" \
              "-DARDUINO_BOARD=\"Espressif ESP32-C3-DevKitM-1\"" -DARDUINO_PARTITION_no_ota \
              -I.pio/libdeps/esp32-c3/NMEA2000_twai -I.pio/libdeps/esp32-c3/NMEA2000-library/src \
              -I/Users/ieb/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32c3/include/newlib/platform_include -I/Users/ieb/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32c3/include/freertos/include -I/Users/ieb/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32c3/include/freertos/include/esp_additions/freertos -I/Users/ieb/.platformio/packages/framework   
          -----snip----- (about 100 other includes)
          spressif32/cores/esp32 -I/Users/ieb/.platformio/packages/framework-arduinoespressif32/variants/esp32c3 .pio/libdeps/esp32-c3/NMEA2000_twai/NMEA2000_esp32.cpp

ttlappalainen, 2024-11-13:
You could also check this https://github.com/offspring/NMEA2000_esp32/tree/add-esp32s3 . I have not yet had time to test it and do merge.

ttlappalainen, 2024-11-13:
Forgot to mention that twai is not in old framework version. So dependencies may be tested according to framework version not just by ESP32 model.

skarlsson, 2024-11-13:
I'm not aware of any differences in 5.x version for twai but I have been running 5.3. Also on many other places I've resorted to C++ style init - not C. ie twai_message_t message = {}; and then assigning individual members - so maybe this is the way here as well. for the record - I was hoping @ttlappalainen would pick this up since I've been jumping in and out of can related things over the years but never stayed for maintenance.

ieb, 2024-11-15:
> You could also check this https://github.com/offspring/NMEA2000_esp32/tree/add-esp32s3

Tried this one, but I am using a C3 not a S3. Also tried to create a C3 header file for that, but all sorts of data structures were different or missing which is why I went for the twai version from @skarlsson  which looked cleaner. I have that loaded and running on a bare board, just getting a pcb cut to test with a driver on a bus. Will report back.

ieb, 2024-11-18:
Ran for > 24h with no failures or painics, although the esp32C3 is single core so race conditions are less likely. Seems to be stable for me.

titio72, 2025-01-06:
I'm trying to use @skarlsson version, but I get compile errors, specifically:
In file included from src/NMEA2000_esp32.cpp:1:
src/NMEA2000_esp32.h:47:5: error: 'twai_handle_t' does not name a type; did you mean 'twai_mode_t'?

I'm using PlatformIO and espressif 32 framework 6.4.0.

Any ideas?

skarlsson, 2025-01-06:
I'm not using platformio - just esp-idf. I assume you mean esp-idf 5.4 - I just bumped to that version and it compiles fine.

titio72, 2025-01-06:
@skarlsson I guess  the difference is that I'm using Arduino rather than esp-idf.

Oscarwaterski, 2025-03-04:
hello everybody, 

is it possible to runit with ARDUINO IDE ?

GWS65, 2025-03-05:
Hi Oscar,

I have used the library for multiple project successfully in the Arduino
IDE v 1.8.19 up to appr a year ago for ESP32 and M5Stack projects. No
experiences with the S2 S3 en C3 chips but I would expect them to work fine
too.

Best regards,
GWS

matthias-mw, 2025-03-09:
Hi @ttlappalainen 
> You could also check this https://github.com/offspring/NMEA2000_esp32/tree/add-esp32s3

Just want to let you know, that this code ist working perfectly fine for me. 

I am using a Seeed Studio XIAO ESP32s3 board with PlatformIO and Arduino Framework.

Here is my platformio.ini
`[env:seeed_xiao_esp32s3]
platform = https://github.com/platformio/platform-espressif32.git
board = seeed_xiao_esp32s3
framework = arduino
monitor_speed = 115200
debug_tool = esp-prog
debug_init_break = tbreak stetup`

This Arduino-Framework uses ESP-IDF Version: v4.4.7.

Thanks for the excellent work.

magerbak, 2025-10-18:
Wanted to add a note that I tested https://github.com/offspring/NMEA2000_esp32/tree/add-esp32s3 on 
• Adafruit ESP32-S3 Reverse TFT Feather (using GPIO pins 5 and 6).
• Arduino 2.3.6
• Espressif IDF release 5.5
and this appears to be working fine. Thanks for adding the S3 support.