ttlappalainen/NMEA2000_esp32 #26: Conflicting Declaration with tNMEA2000_esp32 NMEA2000;

atestani, 2024-06-06:
I am porting a Teensy3.2 NMEA2000 application to ESP32.  I'm using PlatformIO and an ESP32-WROOM-32E.  

platformio.ini is:
```
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
lib_deps = 
	ttlappalainen/NMEA2000-library@^4.21.5
	ttlappalainen/NMEA2000_esp32@^1.0.3
....

```
The start of the code looks like this:
```
// Set up your NMEA2000 CAN pins
#define ESP32_CAN_RX_PIN GPIO_NUM_7
#define ESP32_CAN_TX_PIN GPIO_NUM_4

#include <Arduino.h>
#include <NMEA2000_CAN.h>  // This will automatically include NMEA2000 library
#include <NMEA2000_esp32.h>  
...

// Create NMEA2000 instance
tNMEA2000_esp32 NMEA2000;
```

The last line is giving a conflicting declaration error.  I have searched throughout the code several times and cannot find why this is happening.  Also, I have been unable to find a documented example for use of these libraries on an ESP32.  Any advice?

Thanks
```

```

atestani, 2024-06-07:
I found more information from the error in the build output:
```
In file included from src/main.cpp:39:
.pio/libdeps/esp32dev/NMEA2000-library/src/NMEA2000_CAN.h:278:12: note: previous declaration as 'tNMEA2000& NMEA2000'
 tNMEA2000 &NMEA2000=*(new tNMEA2000_esp32()); 
```

So I don't need to declare the main object?

ttlappalainen, 2024-06-07:
if you use
#include <NMEA2000_CAN.h>  // This will automatically include NMEA2000 library
you do not explicitely include specific MCU driver and define NMEA2000 object.

atestani, 2024-06-07:
Do I still use  #include <NMEA2000_esp32.h>?

ttlappalainen, 2024-06-07:
No. When you use #include <NMEA2000_CAN.h>, it includes necessary driver, if it can. If it can not then you do not #include <NMEA2000_CAN.h> and you have to explicitely include driver and define NMEA2000 object.

atestani, 2024-06-07:
Thanks... good to know.   Is this documented somewhere?  I read everything
I could find before posting here.

atestani, 2024-06-07:
Actually, I found it in the main document under:
   NMEA2000_CAN.h File Reference